gotthardp/python-mercuryapi #93: LLRP Reader Connection attempt is failed

BrunoGiubilei, 2019-11-06:
hey hi, i try to use the sample in README and i get this error.

> Traceback (most recent call last):
>   File "./leitor.py", line 3, in <module>
>     reader = mercury.Reader("tmr://192.198.0.139")
> TypeError: LLRP Reader Connection attempt is failed

above, its the code im try to execute

#! /usr/bin/python
import mercury
reader = mercury.Reader("tmr://192.198.0.139")

reader.set_region("EU3")
reader.set_read_plan([1], "GEN2")
print(reader.read())

gotthardp, 2019-11-06:
Well, this looks like a networking issue. Can you `ping` that IP address?

BrunoGiubilei, 2019-11-06:
@gotthardp yes

> ping 192.168.0.139
> PING 192.168.0.139 (192.168.0.139) 56(84) bytes of data.
> 64 bytes from 192.168.0.139: icmp_seq=1 ttl=255 time=0.881 ms
> 64 bytes from 192.168.0.139: icmp_seq=2 ttl=255 time=0.615 ms
> 64 bytes from 192.168.0.139: icmp_seq=3 ttl=255 time=0.692 ms

BrunoGiubilei, 2019-11-06:
can i debug the script?

gotthardp, 2019-11-06:
Not the script itself, but you can use other tools to check if they can connect or not.

BrunoGiubilei, 2019-11-06:
i check connection using a java binary and i get this error

[image]

gotthardp, 2019-11-07:
This may be due to a firewall, or you need to activate/enable the reader to listen to external devices.

BrunoGiubilei, 2019-11-07:
its works! my reader needs to configure a virtual port to work, i use socat to create a virtual port
> socat pty,link=/dev/ttyV0,waitslave tcp:192.168.0.139:8081

and change the URI in code to my virtual port

[image]

closed 2019-11-07 by BrunoGiubilei

BrunoGiubilei, 2019-11-08:
For anyone who give that same error, or needs to implement over a estructure similiar, i create a gist with two configurations of socat. The code below thats not my final solution, please guide with gist above.

[https://gist.github.com/BrunoGiubilei/474b5933aeff12f574e42477a8e56759](url)