gotthardp/python-mercuryapi #94: Invalid action - GEN2 Select filter

emilesabatier, 2019-11-06:
Hi,

In the GEN2 Select filter Dict, when I set 'action':'on&off' or any other action, the shell displays a "TypeError : invalid action". Do you know why ?

I also have another issue when I put more than 3 epc in a list for the "epc_target" parameter in the "set_read_plan" function, The reader doesn't seem to read anything... 

Thanks a lot :)

gotthardp, 2019-11-06:
You have Python 2.7?

emilesabatier, 2019-11-13:
No I'm using python 3.7

emilesabatier, 2019-11-18:
after my reader init, I have

epcList=[b'E28011700000020D2E970C90',b'E28011700000020D2E970C91',b'E28011700000020D2E970C92',b'E28011700000020D2E970C93',b'E28011700000020D2E970C94',b'E28011700000020D2E970C95',b'E28011700000020D2E970C96',b'E28011700000020D2E970C97',b'E28011700000020D2E970C98',b'E28011700000020D2E970C99']

reader.set_read_plan(antennas, protocol="GEN2", epc_target=epc, read_power=powers[0])

with 
antennas = [1]
powers = [3150]

gotthardp, 2020-08-02:
Hello. While doing a commit c1a18fec251abffab4fe3d4678b8c0490c8eac21 I discovered there was a bug causing the "Invalid action" error. You may want to check the latest master if you still need this.