gotthardp/python-mercuryapi #97: Unable to install on Rasbian

Mixownt, 2019-11-25:
I am unable to install python-mercuryapi both using pip(3) and the manual build. 

Error log: 

> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting python-mercuryapi
>   Using cached https://files.pythonhosted.org/packages/7b/11/f618c649b57145e1b03998cf4d732dedf2d9342a3ffbcbae2ba9bcb8a01c/python-mercuryapi-0.5.2.tar.gz
> Building wheels for collected packages: python-mercuryapi
>   Running setup.py bdist_wheel for python-mercuryapi ... error
>   Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-a_2tlx2l/python-mercuryapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-d16_899y --python-tag cp37:
>   running bdist_wheel
>   running build
>   wget https://www.jadaktech.com/wp-content/uploads/2018/11/mercuryapi-1.31.2.zip
>   --2019-11-25 11:57:38--  https://www.jadaktech.com/wp-content/uploads/2018/11/mercuryapi-1.31.2.zip
>   Resolving www.jadaktech.com (www.jadaktech.com)... 192.124.249.53
>   Connecting to www.jadaktech.com (www.jadaktech.com)|192.124.249.53|:443... connected.
>   HTTP request sent, awaiting response... 404 Not Found
>   2019-11-25 11:57:39 ERROR 404: Not Found.
>   
>   make: *** [Makefile:28: mercuryapi-1.31.2.zip] Error 8
>   running build_ext
>   building 'mercury' extension
>   creating build
>   creating build/temp.linux-armv7l-3.7
>   arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ibuild/mercuryapi/include -I/usr/include/python3.7m -c mercury.c -o build/temp.linux-armv7l-3.7/mercury.o
>   mercury.c:23:10: fatal error: tm_reader.h: No such file or directory
>    #include <tm_reader.h>
>             ^~~~~~~~~~~~~
>   compilation terminated.
>   error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
>   
>   ----------------------------------------
>   Failed building wheel for python-mercuryapi
>   Running setup.py clean for python-mercuryapi
> Failed to build python-mercuryapi
> Installing collected packages: python-mercuryapi
>   Running setup.py install for python-mercuryapi ... error
>     Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-a_2tlx2l/python-mercuryapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-o0knd1ku/install-record.txt --single-version-externally-managed --compile:
>     running install
>     running build
>     wget https://www.jadaktech.com/wp-content/uploads/2018/11/mercuryapi-1.31.2.zip
>     --2019-11-25 11:57:41--  https://www.jadaktech.com/wp-content/uploads/2018/11/mercuryapi-1.31.2.zip
>     Resolving www.jadaktech.com (www.jadaktech.com)... 192.124.249.53
>     Connecting to www.jadaktech.com (www.jadaktech.com)|192.124.249.53|:443... connected.
>     HTTP request sent, awaiting response... 404 Not Found
>     2019-11-25 11:57:42 ERROR 404: Not Found.
>     
>     make: *** [Makefile:28: mercuryapi-1.31.2.zip] Error 8
>     running build_ext
>     building 'mercury' extension
>     creating build
>     creating build/temp.linux-armv7l-3.7
>     arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ibuild/mercuryapi/include -I/usr/include/python3.7m -c mercury.c -o build/temp.linux-armv7l-3.7/mercury.o
>     mercury.c:23:10: fatal error: tm_reader.h: No such file or directory
>      #include <tm_reader.h>
>               ^~~~~~~~~~~~~
>     compilation terminated.
>     error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
>     
>     ----------------------------------------
> Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-a_2tlx2l/python-mercuryapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-o0knd1ku/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-a_2tlx2l/python-mercuryapi/

Mixownt, 2019-11-25:
It seems that the MercuryAPI package is no longer located at  https://www.jadaktech.com/wp-content/uploads/2018/11/mercuryapi-1.31.2.zip. I tried modifying the MakeFile to point to an alternate internet adress (https://www.jadaktech.com/resources/rfid-document-library/thingmagic-mercury-api-v1-9-1-86/?download), but then the updated API seems to generate errors. I attached updated Makefile: 

> make -C mercuryapi-THOREAU-1.9.1.86/c/src/api
> make[1]: Entering directory '/home/pi/python-mercuryapi/mercuryapi-THOREAU-1.9.1.86/c/src/api'
> make[1]: Nothing to be done for 'all'.
> make[1]: Leaving directory '/home/pi/python-mercuryapi/mercuryapi-THOREAU-1.9.1.86/c/src/api'
> mkdir -p build/mercuryapi/include
> find mercuryapi-*/c/src/api -type f -name '*.h' ! -name '*_imp.h' ! -path '*ltkc_win32*' -exec cp {} build/mercuryapi/include/ \;
> mkdir -p build/mercuryapi/lib
> find mercuryapi-*/c/src/api -type f \( -name '*.a' -or -name '*.so.1' \) -exec cp {} build/mercuryapi/lib/ \;
> /usr/bin/python3 setup.py build
> running build
> make[1]: Entering directory '/home/pi/python-mercuryapi'
> make -C mercuryapi-THOREAU-1.9.1.86/c/src/api
> make[2]: Entering directory '/home/pi/python-mercuryapi/mercuryapi-THOREAU-1.9.1.86/c/src/api'
> make[2]: Nothing to be done for 'all'.
> make[2]: Leaving directory '/home/pi/python-mercuryapi/mercuryapi-THOREAU-1.9.1.86/c/src/api'
> mkdir -p build/mercuryapi/include
> find mercuryapi-*/c/src/api -type f -name '*.h' ! -name '*_imp.h' ! -path '*ltkc_win32*' -exec cp {} build/mercuryapi/include/ \;
> mkdir -p build/mercuryapi/lib
> find mercuryapi-*/c/src/api -type f \( -name '*.a' -or -name '*.so.1' \) -exec cp {} build/mercuryapi/lib/ \;
> make[1]: Leaving directory '/home/pi/python-mercuryapi'
> running build_ext
> building 'mercury' extension
> arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ibuild/mercuryapi/include -I/usr/include/python3.7m -c mercury.c -o build/temp.linux-armv7l-3.7/mercury.o
> mercury.c:41:5: error: unknown type name ‘TMR_StatsListenerBlock’
[Makefile.txt](https://github.com/gotthardp/python-mercuryapi/files/3886992/Makefile.txt)

gotthardp, 2019-11-25:
Hello. I fixed the Makefile to use *some* version that can be downloaded. Note it is older than the version used before.
I believe Jadaktech have a bug in their website as the version 1.31.3 downloads as 1.31.1.36-2.

Mixownt, 2019-11-26:
Thank you, I managed to install the wrapper now. However, when running the test.py program, I get TypeError: Timeout. Any ideas as to how this happens? I do not get this problem when initialising the pins with Serial.

Mixownt, 2019-11-26:
Running rebootReader.c also gives "Error connecting reader: Timeout"

gotthardp, 2019-11-26:
Timeout usually means a connectivity issue-- your reader is not connected, not enabled, not configured properly or something along those lines.

Mixownt, 2019-11-28:
I solved the connectivity issues it was indeed on my side, as you suggested!

closed 2019-11-28 by Mixownt

gotthardp, 2019-11-28:
Cool. What was the issue, please? Just in case others have a similar one.

mdjong1, 2019-12-03:
Also interested

Mixownt, 2019-12-03:
> Cool. What was the issue, please? Just in case others have a similar one.

I am using the [Sparkfun ](https://www.sparkfun.com/products/14066)board, which boosts the Tx and Rx signals to 5V to be compatible with Arduino boards. I missed this, because the M6e Nano works at 3.3V and on the Sparkfun product page, they link the M6e datasheet. 

My RPi however works at 3.3V, so I had to put a logic level convertor in between or work via a FTDI breakout to go to USB. It works fine now, but on RPi I frequently get timout or segmentation fault errors, while on my main laptop, I have no issue whatsoever.

clickworkorange, 2023-02-12:
The updated download location seems to be gone now as well - in fact I cannot find anywhere to download the MercuryAPI any more. Installing `python-mercuryapi` with pip also fails on downloading. I do have a copy of mercuryapi-1.31.0.33 from a previous project but this fails to build with 
````
make -C mercuryapi-1.31.0.33/c/src/api
make[1]: Entering directory '/home/myself/Library/ThingMagic M6e RFID reader/python-mercuryapi/mercuryapi-1.31.0.33/c/src/api'
cc -Ilib/LTK/LTKC/Library -Ilib/LTK/LTKC/Library/LLRP.org -I. -g -Wall -fPIC -shared -Wl,-rpath,/tm/lib,-soname,libmercuryapi.so.1 -o libmercuryapi.so.1 serial_transport_posix.o serial_transport_tcp_posix.o tmr_strerror.o tmr_param.o hex_bytes.o tm_reader.o tm_reader_async.o serial_reader.o tmr_loadsave_configuration.o llrp_reader.o llrp_reader_l3.o serial_reader_l3.o tmr_utils.o osdep_posix.o  -lpthread
/usr/bin/ld: serial_transport_posix.o: relocations in generic ELF (EM: 40)
/usr/bin/ld: serial_transport_posix.o: relocations in generic ELF (EM: 40)
/usr/bin/ld: serial_transport_posix.o: relocations in generic ELF (EM: 40)
/usr/bin/ld: serial_transport_posix.o: relocations in generic ELF (EM: 40)
/usr/bin/ld: serial_transport_posix.o: relocations in generic ELF (EM: 40)
/usr/bin/ld: serial_transport_posix.o: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:154: libmercuryapi.so.1] Error 1
make[1]: Leaving directory '/home/myself/Library/ThingMagic M6e RFID reader/python-mercuryapi/mercuryapi-1.31.0.33/c/src/api'
make: *** [Makefile:13: mercuryapi] Error 2
````
This is on Linux 6. It seems some serial device descriptor thingy (mumble-mumble) has changed, but searching the Intertubes for the error `/usr/bin/ld: serial_transport_posix.o: error adding symbols: file in wrong format` turns up zilch, zip, nada - which is always a worrying sign.