gotthardp/python-mercuryapi #98: Hop time issue (switching every 250 ms)

gaHog, 2019-11-26:
Hello,

Thanks a lot for all the good work on this wrapper.

I have an issue when I try to set the hop_time, whether I set it at 3975ms or 2000 ms, it seems like it stays set at roughly 250 ms, here are the code I use and results I get :

Code : 

import mercury
import time
from datetime import datetime

reader = mercury.Reader("tmr:///dev/ttyACM0")
reader.set_region("EU3")
reader.set_read_plan([1], "GEN2", read_power = 1200)
reader.set_hop_table([865700, 866300, 866900, 867500])
reader.set_gen2_session(0)
reader.set_gen2_target(0)
reader.set_gen2_q(1,1)

print(reader.get_hop_table(), reader.set_hop_time(2000))    #or 3975

reader.start_reading(lambda tag: print(tag.epc, tag.rssi, datetime.fromtimestamp(tag.timestamp), tag.frequency,  tag.antenna))
time.sleep(1.1)
reader.stop_reading()

Results :

[865700, 866300, 866900, 867500] 2000
b'E28011700000020B0665200D' -39 2019-11-26 14:51:06.970000 866900 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:06.996000 866900 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.022000 866900 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.048000 866900 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.073000 866900 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.099000 866900 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.124000 866900 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.151000 866900 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.176000 866900 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.201000 866900 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.233000 867500 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.258000 867500 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.285000 867500 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.310000 867500 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.336000 867500 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.362000 867500 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.386000 867500 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.413000 867500 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.438000 867500 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.464000 867500 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.483000 865700 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.508000 865700 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.534000 865700 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.560000 865700 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.586000 865700 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.612000 865700 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.637000 865700 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.663000 865700 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.688000 865700 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.714000 865700 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.734000 866300 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.760000 866300 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.787000 866300 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.812000 866300 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.837000 866300 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.863000 866300 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.888000 866300 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.915000 866300 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.940000 866300 1
b'E28011700000020B0665200D' -39 2019-11-26 14:51:07.965000 866300 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:07.986000 866900 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:08.011000 866900 1
b'E28011700000020B0665200D' -40 2019-11-26 14:51:08.037000 866900 1

[865700, 866300, 866900, 867500] 3975
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.359000 866900 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.384000 866900 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.410000 866900 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.436000 866900 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.462000 866900 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.487000 866900 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.512000 866900 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.539000 866900 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.564000 866900 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.589000 866900 1
b'E28011700000020B0665200D' -45 2019-11-26 14:40:57.622000 867500 1
b'E28011700000020B0665200D' -45 2019-11-26 14:40:57.648000 867500 1
b'E28011700000020B0665200D' -45 2019-11-26 14:40:57.673000 867500 1
b'E28011700000020B0665200D' -45 2019-11-26 14:40:57.698000 867500 1
b'E28011700000020B0665200D' -45 2019-11-26 14:40:57.724000 867500 1
b'E28011700000020B0665200D' -45 2019-11-26 14:40:57.750000 867500 1
b'E28011700000020B0665200D' -45 2019-11-26 14:40:57.775000 867500 1
b'E28011700000020B0665200D' -45 2019-11-26 14:40:57.802000 867500 1
b'E28011700000020B0665200D' -45 2019-11-26 14:40:57.827000 867500 1
b'E28011700000020B0665200D' -45 2019-11-26 14:40:57.852000 867500 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.873000 865700 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.899000 865700 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.925000 865700 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.949000 865700 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:57.976000 865700 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.001000 865700 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.027000 865700 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.052000 865700 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.079000 865700 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.104000 865700 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.126000 866300 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.151000 866300 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.178000 866300 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.204000 866300 1
b'E28011700000020B0665200D' -42 2019-11-26 14:40:58.230000 866300 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.255000 866300 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.281000 866300 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.307000 866300 1
b'E28011700000020B0665200D' -44 2019-11-26 14:40:58.332000 866300 1

Maybe I am doing something wrong ? If possible I would like to set the hop_time to around 10 seconds, would it be possible ?

Thanks in advance for your help !

gotthardp, 2019-11-26:
Hello. This is a reader reader related problem, so I cannot help help much.
Some tips though:
Sometimes, it helps to change order of the commands, i.e. set hop time before/after the hop table and/or set both before/after setting the read plan. Have you tried that?
Also, sometimes there are some restrictions documented in the MercuryAPI and/or reader documentation. Have you tried other values?

gaHog, 2019-11-27:
Hello, 
I tried what you suggested, changing the order of the commands did not change a thing...

Moreover, even when I don't set the hop time, nor the hop table (initialized at  3975ms and [865700, 866300, 866900, 867500] for EU frequencies), the results I get remain the same, there are frequency hop every 250 ms...

I checked the mercuryAPI as well, it doesn't seem like the hop_time value is to be selected among a list of available values. here is what I read :

"Single Channel Mode
• Set by manually setting the frequency hop table to a single frequency. In this mode the module will occupy the set channel for up to four seconds, after which it will be quiet for 100ms before transmitting on the same channel again.

Multi Channel Mode
• Set by leaving the default or manually setting more than one frequency in the hop table. In this mode the module will occupy one of the configured channels for up to four seconds, after which it may switch to another channel and immediately occupy that channel for up to four seconds. This mode allows for continuous operation."

Changing the value from 250 to 3975 didn't change anything to the results. but, when I set it on 100ms, I could see the frequency hop happen every 100 ms

gaHog, 2020-06-02:
Hello,

I discussed this issue with Thingmagic and they solved it a while ago.
You were right,  it was a reader related issue.
I forgot to close this issue afterwards, so I do it now,

Thanks a lot for your help !

closed 2020-06-02 by gaHog