gotthardp/python-mercuryapi #99: Unable to install on Raspberry PI

maalliet, 2019-12-02:
Hello,

I'm unable to download the python-mercury library following the two installation options on the Raspberry PI.

pip install returns following output (seems that library on jadaktech has been moved?)

`user@raspberrypi:~ $ sudo pip install python-mercuryapi
Collecting python-mercuryapi
  Using cached https://files.pythonhosted.org/packages/7b/11/f618c649b57145e1b03998cf4d732dedf2d9342a3ffbcbae2ba9bcb8a01c/python-mercuryapi-0.5.2.tar.gz
Building wheels for collected packages: python-mercuryapi
  Running setup.py bdist_wheel for python-mercuryapi ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Tw7qGF/python-mercuryapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpn3vdIopip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  wget https://www.jadaktech.com/wp-content/uploads/2018/11/mercuryapi-1.31.2.zip
  --2019-12-02 20:18:05--  https://www.jadaktech.com/wp-content/uploads/2018/11/mercuryapi-1.31.2.zip
  Redirect from www.jadaktech.com (www.jadaktech.com)... 192.124.249.53
  Connecting to www.jadaktech.com (www.jadaktech.com)|192.124.249.53|:443... connected.
  HTTP-request is sent; waiting for response... 404 Not Found
  2019-12-02 20:18:07 Fout 404: Not Found.

  Makefile:28: recipe for target 'mercuryapi-1.31.2.zip' has failed
  make: *** [mercuryapi-1.31.2.zip] Error 8
  running build_ext
  building 'mercury' extension
  creating build
  creating build/temp.linux-armv7l-2.7
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-Ub3vap/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Ibuild/mercuryapi/include -I/usr/include/python2.7 -c mercury.c -o build/temp.linux-armv7l-2.7/mercury.o
  mercury.c:23:23: fatal error: tm_reader.h: File or directory does not exist
   #include <tm_reader.h>
                         ^
  compilation terminated.
  error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

  ----------------------------------------
  Failed building wheel for python-mercuryapi
  Running setup.py clean for python-mercuryapi
Failed to build python-mercuryapi
Installing collected packages: python-mercuryapi
  Running setup.py install for python-mercuryapi ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Tw7qGF/python-mercuryapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-qEuTn3-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    wget https://www.jadaktech.com/wp-content/uploads/2018/11/mercuryapi-1.31.2.zip
    --2019-12-02 20:18:10--  https://www.jadaktech.com/wp-content/uploads/2018/11/mercuryapi-1.31.2.zip
    Redirect from www.jadaktech.com (www.jadaktech.com)... 192.124.249.53
    Connecting to www.jadaktech.com (www.jadaktech.com)|192.124.249.53|:443... connected.
    HTTP-request is sent; waiting for reply... 404 Not Found
    2019-12-02 20:18:10 Fout 404: Not Found.

    Makefile:28: recipe for target 'mercuryapi-1.31.2.zip' has failed
    make: *** [mercuryapi-1.31.2.zip] Fout 8
    running build_ext
    building 'mercury' extension
    creating build
    creating build/temp.linux-armv7l-2.7
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-Ub3vap/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Ibuild/mercuryapi/include -I/usr/include/python2.7 -c mercury.c -o build/temp.linux-armv7l-2.7/mercury.o
    mercury.c:23:23: fatal error: tm_reader.h: File or directory does not exist
     #include <tm_reader.h>
                           ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Tw7qGF/python-mercuryapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-qEuTn3-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Tw7qGF/python-mercuryapi/`

The manual approach allows me to install the mercury api without any visible errors.
Although, when running a script (tried both python3 and python2), an Import Error is thrown:

`ImportError: No module named mercury`

The RPI OS is the following:
`PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

gotthardp, 2019-12-03:
Hello. True, the download location changed. That's beyond my control.

This is duplicate of #97. I changed the links some days ago, but didn't release the pip package. I fixed that too now.

closed 2019-12-03 by gotthardp

stevennoyce, 2020-03-17:
I've seen a few issues relating to the inability to install due to missing files.  I'm not sure what the underlying theme is that causes this to be a repeated issue that keeps getting fixed, but I'm having currently unable to install on a raspberry pi via pip.

Currently I am missing the file `tm_reader.h`:

````
mercury.c:23:10: fatal error: tm_reader.h: No such file or directory
#include <tm_reader.h>
         ^~~~~~~~~~~~~
compilation terminated.
error: command 'gcc' failed with exit status 1
````

I have tried working back through older versions on pip to see if any of them are able to install, but all are missing at least one file (different files for different versions).

Any advice or a fix would be highly appreciated.

Thanks.