gotthardp/python-mercuryapi #100: Getting TypeError: Timeout

mdjong1, 2019-12-03:
Getting error "TypeError: Timeout" when using ```reader = mercury.Reader("tmr:///dev/ttyUSB0", baudrate=9600)```

Not getting much else debug info; any ideas?

mdjong1, 2019-12-03:
Looks like this is related to (TTL) connectivity issues on our end

closed 2019-12-03 by mdjong1

jacepetersen12, 2019-12-20:
@mdjong1  Just curious what issues you had. I'm having a similar problem. I can get the reader to work on my raspberry pi 3 b+ but no my Zero w. Timeout error on that same line

mdjong1, 2019-12-22:
We ended up having issues with the USB TTL cable which was faulty. Weren't able to read from any device using any library, so finally figured out what it was.

Headforthehills, 2020-08-18:
@jacepetersen12 Did you solve the timeout error on the zero? I too have a working project on a 3b+ that I need to run on a Pi Zero. I get the "TypeError: Timeout" on the zero only.