gotthardp/python-mercuryapi #102: read TID tag using python

ramzimosalam, 2020-02-06:
**i am  use this code:**

reader = mercury.Reader("tmr:///dev/ttyUSB0", baudrate=115200)

print(reader.get_model())
# print(reader.get_supported_regions())

reader.set_region("EU3")
reader.set_read_plan([1], "GEN2",bank=["tid"] ,read_power=1900)
# print(reader.read())

reader.start_reading(lambda tag: print(tag.tid, tag.antenna, tag.read_count, tag.rssi))
time.sleep(1)
**when i execute the code i receive this message:**
Traceback (most recent call last):
  File "test.py", line 54, in <lambda>
    reader.start_reading(lambda tag: print(tag.tid, tag.antenna, tag.read_count, tag.rssi))
AttributeError: 'mercury.TagReadData' object has no attribute 'tid'

**can anyone help me,thank.**

gotthardp, 2020-02-07:
Which python-mercuryapi version do you use?

markwaynelea, 2020-11-20:
replace tag.tid in your print statement with tag.tid_read_data since the TagReadData object has no tid attribute.