gotthardp/python-mercuryapi #109: start_reading callback never gets called if run inside new thread?

clickworkorange, 2020-06-24:
My multiprocessing fu is weak, and I cannot figure out why the start_reading callback ("found") never gets called here: 
````python
#!/usr/bin/env python3

from mercury import Reader
from multiprocessing import Process
from time import sleep

class MyProcess(Process):
  def __init__(self):
    Process.__init__(self)
    self.reader = Reader("tmr:///dev/ttyAMA0", baudrate=115200)

  def run(self):
    print("Run")
    self.reader.start_reading(self.found)

  def stop(self):
    print("Stop")
    self.reader.stop_reading()

  def found(self, tags):
    print(tags)

if __name__ == '__main__':
  try:
    process = MyProcess()
    process.daemon = True
    process.start()
    while True:
      sleep(0.1)
    
  except KeyboardInterrupt:
    process.stop()
````
Is it because `self.found()` doesn't point where I think it does? I don't get any reference errors though, or any other error - but the `found` never gets called. Any ideas?

HanYangZhao, 2020-06-29:
Try this:
```
  def run(self):
    print("Run")
    self.reader.start_reading(self.found)
    while(True):
        sleep(1)
```
If you're not doing `while(True)`, then the run function finishes executing and the process terminates, leaving you with just the main process in sleep(0.1)

clickworkorange, 2020-06-30:
Thanks! That's interesting. I knew the function would terminate, but not that this would take with it the reader and/or the callback. I thought of run() as being tasked with starting the reader and then getting out of the way, since there is no more work for it to do. Doesn't __init__ terminate too?

clickworkorange, 2020-06-30:
Actually, never mind - this isn't the right place for such questions!

closed 2020-06-30 by clickworkorange