gotthardp/python-mercuryapi #113: Gen2 Select Filtering

tberndt5, 2020-06-30:
Hello,
I am trying to use the gen2 select filtering as described in the documentation, but I keep getting "TypeError:reserved, epc, tid or user mask expected". I am not sure what I am doing wrong. Is there anyone that could present an example of the gen2 select filtering with all of the associated items in the dict. 

Thanks

n8cutler, 2020-07-14:
Hi,

I'm also struggling to get the Gen2 Select filtering working and could really use a working example.
I've tried building a dictionary and list of dictionaries according to the fields [defined](https://github.com/gotthardp/python-mercuryapi#target-filtering). 
However, I can't seem to get the format right and I'm sure I'm missing something simple.

```bash
TypeError: byte string expected
TypeError: filter expected
```
Any help would be appreciated. 
Thanks.

amsimon, 2020-09-10:
@n8cutler Did you get this to work?