gotthardp/python-mercuryapi #114: Windows installation

choucous, 2020-07-29:
I’m trying to install the API on Windows 10. The instruction says.
« On Windows, use the pre-compiled binary installer. »
Where can I find this binary installer and could you give me more details about the installation procedure.
Thank you un advance

luchermans, 2020-12-09:
under Assets in v0.5.1
https://github.com/gotthardp/python-mercuryapi/releases/tag/v0.5.1

luchermans, 2020-12-15:
using Python 3.6.8  32bit installed mercuryapi-0.5.1.win32-py3.6.exe
on import mercury i get
ImportError: DLL load failed: The specified module could not be found.
I have installed https://www.microsoft.com/en-us/download/details.aspx?id=5555
but same error

FKPSC, 2022-03-31:
The Microsoft link above is broken, use this one:
https://www.microsoft.com/en-us/download/details.aspx?id=26999

Also, only the 32 bit release worked:
https://github.com/gotthardp/python-mercuryapi/releases/download/v0.5.1/mercuryapi-0.5.1.win32-py3.6.exe

caveman2st, 2022-08-28:
@luchermans, thank you for this info. it took me a long time to find the windows installer.
@gotthardp, I would be happy if there is a new version because i like epcpy but it needs Python >= 3.7