gotthardp/python-mercuryapi #116: Import Error undefined symbol: PyUnicode_AsUTF8

mxmjnn, 2020-07-31:
Hello (disclaimer first post on Github),

I got an import error when trying to use this package with a raspberry pi model 3b with Raspbian: 

ImportError: /home/pi/.local/lib/python2.7/site-packages/mercury.so: undefined symbol: PyUnicode_AsUTF8

Seems similar to the issue described here but was not able to solve this yet:

https://github.com/gotthardp/python-mercuryapi/issues/28

Any ideas to point me in the right direction?

Thanks!

mxmjnn, 2020-07-31:
Quick update after continued trouble shooting:

import was successful when specifying python3 when running python script. Could there be a compatibility issue with python 2? It seems like this package should be able to work on both

gotthardp, 2020-08-02:
Hello. I just made a fix. Please try building the latest master and let me know if it helped.

closed 2020-11-25 by gotthardp