gotthardp/python-mercuryapi #119: Phase data output explanation

andrewruba, 2020-09-23:
Hi! This is a really fun api. I got it working with an older Thingmagic Vega reader I bought on ebay and my raspberry pi. I have a question about the phase of the tag. For me, it seems that the phase values are in the range 0 to 65535. Here is a typical time sequence of readings: [65257, 65473, 65359, 65311, 65290, 65465, 65336, 2080, 65052, 65369, 462, 88, 65239, 526, 265, 65439, 2006, 64968, 36, 544, 65143, 64625, 1593, 65406, 601]

Can someone help me understand what these represent? Is it 0 to pi or 0 to 2pi scaled to a 16 bit range? Is it just an error?

c0r3y123, 2023-01-27:
Do you have your code?

andrewruba, 2023-01-27:
I don't know exactly where it is at the moment (not playing around with this anymore) but I recall only using a few lines of code to connect the reader and read the phase data from TagReadData Object. Nothing fancy.

c0r3y123, 2023-01-27:
Oh ok. I tried using their provided test.py, but it only prints out an empty list. If you have free time, do you mind trying to find it? Thanks!

andrewruba, 2023-01-27:
Your reader might not have phase data to give to the api, just fyi. You can check the docs for your particular model.