gotthardp/python-mercuryapi #132: Installation error due to dead link of Jadak API

ArnoldGitHub, 2022-11-22:
Hi,

When I try to install via `pip install python-mercuryapi` , the following error shows:
`      unzip:  cannot find zipfile directory in one of mercuryapi-1.31.1.36-2.zip or
              mercuryapi-1.31.1.36-2.zip.zip, and cannot find mercuryapi-1.31.1.36-2.zip.ZIP, period.
      make: *** [mercuryapi-1.31.1.36/.done] Error 9`

And when i further check, the link that the installation module try to curl is https://www.jadaktech.com/wp-content/uploads/2019/10/mercuryapi-1.31.1.36-2.zip which doesn't exist.

I know that Jadak has recently did a big revamp of their website in around Oct, 2022.
Now downloading the SDK requires filling in a contact form, and the download link of the zip file was changed without a redirection. It should be the reason.

Arnold

dnloliveira66, 2022-12-03:
Hi, i'm having the same issue due th JADAK's link i believe, here's what it's shown:

Collecting python-mercuryapi
  Using cached python-mercuryapi-0.5.3.tar.gz (21 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: python-mercuryapi
  Building wheel for python-mercuryapi (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [13 lines of output]
      running bdist_wheel
      running build
      'make' nao ‚ reconhecido como um comando interno
      ou externo, um programa oper vel ou um arquivo em lotes.
      running build_ext
      building 'mercury' extension
      creating build
      creating build\temp.win-amd64-cpython-39
      creating build\temp.win-amd64-cpython-39\Release
      "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.34.31933\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Ibuild/mercuryapi/include -IC:\users\daniel\appdata\local\programs\python\python39\include -IC:\users\daniel\appdata\local\programs\python\python39\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.34.31933\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\VS\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\cppwinrt" /Tcmercury.c /Fobuild\temp.win-amd64-cpython-39\Release\mercury.obj
      mercury.c
      mercury.c(23): fatal error C1083: Nao ‚ poss¡vel abrir arquivo incluir: 'tm_reader.h': No such file or directory
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.34.31933\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for python-mercuryapi
  Running setup.py clean for python-mercuryapi
Failed to build python-mercuryapi
Installing collected packages: python-mercuryapi
  Running setup.py install for python-mercuryapi ... error
  error: subprocess-exited-with-error

  × Running setup.py install for python-mercuryapi did not run successfully.
  │ exit code: 1
  ╰─> [15 lines of output]
      running install
      C:\users\daniel\appdata\local\programs\python\python39\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      'make' nao ‚ reconhecido como um comando interno
      ou externo, um programa oper vel ou um arquivo em lotes.
      running build_ext
      building 'mercury' extension
      creating build
      creating build\temp.win-amd64-cpython-39
      creating build\temp.win-amd64-cpython-39\Release
      "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.34.31933\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Ibuild/mercuryapi/include -IC:\users\daniel\appdata\local\programs\python\python39\include -IC:\users\daniel\appdata\local\programs\python\python39\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.34.31933\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\VS\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22000.0\\cppwinrt" /Tcmercury.c /Fobuild\temp.win-amd64-cpython-39\Release\mercury.obj
      mercury.c
      mercury.c(23): fatal error C1083: Nao ‚ poss¡vel abrir arquivo incluir: 'tm_reader.h': No such file or directory
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.34.31933\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> python-mercuryapi

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

Headforthehills, 2022-12-10:
Hi,

I am trying to install on Pi 3B+ and getting the Error 9 also.:

pi@raspberrypi:~/python-mercuryapi $ sudo make install
unzip mercuryapi-YEATS-1.31.4.35-1.zip
Archive:  mercuryapi-YEATS-1.31.4.35-1.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of mercuryapi-YEATS-1.31.4.35-1.zip or
        mercuryapi-YEATS-1.31.4.35-1.zip.zip, and cannot find mercuryapi-YEATS-1.31.4.35-1.zip.ZIP, period.
make: *** [Makefile:23: mercuryapi-1.31.4.35/.done] Error 9

Voikat, 2022-12-25:
Is there someone with a solution for this problem because I have the same and I really need help

pytremb, 2023-03-02:
Look at issue #86. That enabled me to do a successful install this morning.

airlomba, 2023-04-10:
Hi, the Makefile (with wrong links) has been corrected. Replace your Makefile with a fresh one from the repo. As of this date, it works.

lefty01, 2024-03-11:
for what it's worth I got that working with the latest api version (with python 3.12):
```
APIZIP ?= mercuryapi-BILBO-1.37.2.24.zip
APIVER ?= 1.37.2.24
```
but manually downloaded that zip from https://www.jadaktech.com/documents-downloads/thingmagic-mercury-api-1-37-2/?download and copied into the git clone of this repo. then make & make install worked.
looks jadaktech does not have a direct download link for this but requires a registration step. see #142