gotthardp/python-mercuryapi #133: make error

Voikat, 2022-12-25:
Hey, I need to install MercuryAPI on my Raspberry. I followed your instructions but when I use the 'make' command I get 

unzip mercuryapi-YEATS-1.31.4.35-1.zip
Archive:  mercuryapi-YEATS-1.31.4.35-1.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of mercuryapi-YEATS-1.31.4.35-1.zip or
        mercuryapi-YEATS-1.31.4.35-1.zip.zip, and cannot find mercuryapi-YEATS-1.31.4.35-1.zip.ZIP, period.
make: *** [Makefile:23: mercuryapi-1.31.4.35/.done] Error 9

I really need help if I made a mistake or something.
[image]

pytremb, 2023-03-02:
Look at issue #86. That fixed the problem for me.

airlomba, 2023-04-10:
Hi, the Makefile has been corrected. Replace the one you have with the one in the repository. :)