gotthardp/python-mercuryapi #142: Can you please update the 'pip install python-mercuryapi' for the new jadak mercury api?

cor3y123, 2023-09-10:


lefty01, 2024-03-11:
btw. I just came along this project, I was able to manually download that latest api (mercuryapi-BILBO-1.37.2.24.zip) from:
https://www.jadaktech.com/documents-downloads/thingmagic-mercury-api-1-37-2/?download
that required to register some email addr.
however after that with changing there in the makefile
```
APIZIP ?= mercuryapi-BILBO-1.37.2.24.zip
APIVER ?= 1.37.2.24
```
just running make and make installed worked (with some warning see below) ... this was done under Linux Fedora 39 / python 3.12

```
$ make install
make -C mercuryapi-1.37.2.24/c/src/api
...
/usr/bin/python3 setup.py install
running install
/usr/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer, pypa/build or
        other standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************
!!
  self.initialize_options()
/usr/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer, pypa/build or
        other standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************
```

sanjaygorak, 2024-10-29:
are we still able to run this??