gotthardp/python-mercuryapi #144: Using this wrapper on new M7E pico dev kit

TinManAkshay, 2023-12-20:
Hi - 

I've been using this wrapper on m6e nano dev kit for a while now without any issues. Since its gone obsolete now. So I switched to new M7E PICO dev kit which i got from digikey.

I had to change NA2 to NA in reader.set_region(). 

I am not able to read any tags. So I'd like to know if this wrapper can be used on M7E pico? If yes, what changes i have to make in order to read tags? Should I need to update this wrapper on my system for m7e? 

Any feedback or info is highly appreciated.

Thanks,
Akshay

gotthardp, 2023-12-21:
Most likely te MercuryAPI version you are using does not support your hardware. Check the release documentation.

TinManAkshay, 2023-12-21:
Yes I figured that out as well. I was using 1.31.1 version with M6E which will not support latest M7E. So I completely removed 1.31.1 and freshly installed the 1.35.2. This latest rev works pretty well with M7E without any issues. 

Thanks for your feedback though. Appreciate it. Closing this!!

closed 2023-12-21 by TinManAkshay