gotthardp/python-mercuryapi #153: Key not found error

maddy81, 2024-09-23:
I am running this on windows and when i am reading by 

 reader = mercury.Reader("tmr://com5",baudrate=115200,protocol = "GEN2")
 
 it says 
 
 TypeError: Key not found. I have set the baudrate to 115200 in device manager
 
 any solution for this 
 
 thanks in advance.

giovannicanedoli, 2024-09-23:
First of all, as in the documentation you need to add an extra '/' character in your string so that your device will be correctly identified. Secondly, make sure that com5 is correctly plugged to your machine and that baudrate is correct. Hope this helps.

maddy81, 2024-09-23:
> First of all, as in the documentation you need to add an extra '/' character in your string so that your device will be correctly identified. reader = mercury.Reader("tmr:///COM5", baudrate=115200,protocol="GEN2") Secondly, make sure that com5 is correctly plugged to your machine and that baudrate is correct. Hope this helps.

maddy81, 2024-09-23:
Thank you soo much the issue was with the slash and when i put it it works fine. Thank you soo much.

closed 2024-09-23 by maddy81